TalismanDesign/PocketMage_PDA #1: REQUEST: Add Gerber/case fi,les so we can make our own?

sp-rky, 2025-01-17:
Hi! 

I love this project, and would love to build this myself as I'd love to turn it into a little portable SSH terminal! Would you be interested in sharing the files you used to manufacture this? eg. The Gerber files for the PCB, the STL files for the case, and (if you feel up to it, I know it would be a big time investment), a rough build guide/parts list? 

Anyways, thanks for your video, it got the little cogs in my brain turning with all the things you could do with this little gadget :)

Thanks,
Steph

derrickharrell, 2025-02-15:
I too would like to make my own.  I would be very grateful if you posted the gerbers, CAD schematic and 3d model for the case.  Thanks for this cool project idea.

ashtf8, 2025-03-04:
Hello,
The Gerber, p&p, BOM, 3D Files, etc. will become available when the project is mature enough where I am comfortable with releasing it as a complete product. Thank you for your interest and patience!

Ashtf

closed 2025-03-04 by ashtf8