TalismanDesign/PocketMage_PDA #6: things to add perhaps

doggomanic, 2025-06-05:
music/video player for local music/videos
darkmode
webbrowser
usb ports
tf/sd card
mayhaps optional digital display like an lcd or led display
more punctuation such as <>/?;:'"[{]}\|`~!@#$%^&*() up down arrows 
word/line/sentance skip
delete button in addition to backspace
home button

maybe even look at older versions of android for inspiration give it opensource and free nature it might be of help or heaven forbid use ai for the code

theflynn49, 2025-06-24:
and a calc app maybe ?

bormilan, 2025-06-25:
Considering the goal of the project I think adding features like browser or video player is just unnecessary. I'm new to this project/community so maybe I'm not right, but as I understand the vision, this device was invented to be a note taking device, without any distracting app.

closed 2025-07-18 by ashtf8