TalismanDesign/PocketMage_PDA #11: Any chance for a Terminal app?

mildwood, 2025-08-05:
My first tought was that it would be a lot more fun to use this device if we could run a terminal on it to SSH into servers etc...IF it is technically possible I maybe can start working on it in freetime if I will be able to buy a unit.

Jonson26, 2025-08-06:
The project is said to use the ESP32-S3, which has the necessary hardware, it's just that the software does not seem to be using it right now.

Finbear2, 2025-08-07:
Should we display the whole interaction (Command & Output) on the oled one line at a time like the dice rolling app. Or should we make it a standalone app accessible from the homepage that displays the command on the e-ink? Maybe even display the command on the e-ink aswell for readability?

Error-Gap, 2025-09-09:
I noticed that the future features/wishlist includes: 

"WiFi notes/calendar sync (Google Docs/Calendar)"

So I'm guessing that underlying wifi stuff will come with that. Then we just need an SSH/Terminal client

rbrtjns90, 2025-12-09:
I think that's doable. Now that I'm done with the emulator I can take a crack at it

khlsvr, 2025-12-11:
I asked chatgpt about low power alternative to ssh/mosh with wifi by using BLE instead. This would be way more battery-friendly than leaving your wifi on in case you have a companion device like a phone or pc near you. Below is the question and the answer.

ME: how about if I create a terminal app for the pocketmage but instead of it being ssh/mosh connection to a remote server using internet, it would be using BLE that would receive the data from the ble-connected device which in turn would be connected to ssh/mosh server. Would it be possible? I would be updating the screen not constantly, but say once in in 10 seconds, and optionally manually with some key combo?

Chatgpt: 

# BLE-Based Remote Terminal Concept for PocketMage (ESP32 + E-ink)

**Short summary:**  
Yes — it is fully possible to build a low-power terminal on the PocketMage that communicates over **BLE** to a nearby device. The companion device runs SSH/MOSH, while the PocketMage only displays text and sends keystrokes. BLE throughput is enough, e-ink refreshes can be infrequent, and the power consumption is far lower than using Wi-Fi.

<details>
<summary>Full technical explanation</summary>

## 1. High-level feasibility
The architecture:

PocketMage (ESP32 + BLE + E-ink)
⇅ BLE
Phone / Laptop / Companion device
⇅ Internet
SSH / MOSH server

The PocketMage becomes a **low-power display + keyboard**.  
The companion device runs the actual SSH/MOSH client.  
This setup is absolutely doable.

---

## 2. BLE can carry the terminal data
- BLE throughput: typically **20–80 kbit/s** (enough for terminal text).
- Latency: **20–100 ms**, acceptable for typing.
- Terminal screens are tiny (e.g., 80×24 ≈ 2 KB), so diffs are small.

BLE is sufficient for:
- Full screen refreshes
- Incremental diffs
- Keystroke events

---

## 3. Power comparison: BLE vs Wi-Fi
- BLE (low duty): **sub-1 mA average** with good connection intervals.
- Wi-Fi idle: **20–50 mA**.
- Wi-Fi TX peaks: **150–250 mA**.

A BLE-based terminal with e-ink and infrequent updates can achieve **days to weeks** of battery life.  
A Wi-Fi-based SSH client on the ESP32 typically lasts **hours to a day**.

---

## 4. Recommended architecture

### **On PocketMage (ESP32):**
Implement a custom BLE GATT service:

- **RX characteristic (to PocketMage):**  
  Receives terminal diffs or full-frame updates.
- **TX characteristic (from PocketMage):**  
  Sends keystrokes back.
- Optional **control characteristic** for commands like full e-ink refresh.

### **On the companion device:**
A small daemon that:

1. Connects over BLE.
2. Runs SSH or MOSH locally.
3. Captures terminal output.
4. Converts updates to diffs.
5. Sends them to PocketMage.
6. Applies keystrokes received from BLE.

This offloads all cryptography, networking, and heavy logic.

---

## 5. E-ink refresh strategy
Your plan is ideal:

- Refresh **every ~10 seconds** automatically for idle dashboards.
- Manual refresh via key combo for reading or interactive use.
- Partial refresh if your display supports it.

This keeps power consumption very low.

---

## 6. Practical limitations
This approach is **not** suited for:

- Real-time animations
- Very rapidly updating curses apps (e.g., htop at 200 ms refresh)

It works excellently for:

- CLI sessions
- System administration
- Reading logs
- Git operations
- Occasional editing

---

## 7. Security model
- SSH/MOSH encryption stays on the **companion device**.
- BLE only transports rendered terminal text + keystrokes.
- You can encrypt BLE payloads if you want, but not strictly necessary.

---

## 8. Conclusion
This is a **valid**, **low-power**, and **clean** design for PocketMage:

- BLE gives huge battery savings.
- Terminal data fits within BLE bandwidth comfortably.
- E-ink allows infrequent, cheap screen updates.
- The architecture mirrors classic serial terminals.

If needed, I can provide:
- GATT service definitions  
- Diffing strategies  
- ESP32 BLE parameter recommendations  
- Pseudocode for both sides  

</details>

ashtf8, 2026-02-08:
Implemented as of Beta V1.5

closed 2026-02-08 by ashtf8