TalismanDesign/PocketMage_PDA #13: Virtual Hardware for development

terr-steak, 2025-08-08:
Hi I just discovered your project and was looking to contribute. I just signed up for the waitlist on the HW but in the meantime I was wondering if you know of anyway to setup a virtual environment to work with in the mean time? I am new to the hardware side of things and will be looking into it myself and will update this thread with what I find, but in the meantime I figured I would ask here as well.

arthurjbc, 2025-08-19:
I couldnt get it to work on linux either

[image]

isnt cocoa a macOS tech?

Jonson26, 2025-08-19:
Changing line 231 in HOME.cpp to `if ((currentMillis*1.0 - OLEDFPSMillis) >= (1000.0/OLED_MAX_FPS)) {` makes the main menu work somewhat, however there are a few more issues:
1. It seems that the font for the OLED is not loaded, so nothing is displayed in the OLED window
2. More floating point related crashes occur when entering apps. I suspect similar fixes could help

rbrtjns90, 2025-08-19:
> Changing line 231 in HOME.cpp to `if ((currentMillis*1.0 - OLEDFPSMillis) >= (1000.0/OLED_MAX_FPS)) {` makes the main menu work somewhat, however there are a few more issues:
>
> 1. It seems that the [...]

[image]

I've got it working in Windows. Let me check Linux.

Jonson26, 2025-08-20:
I think part of the problem on linux might be the availability of fonts. It might be easier to bundle them with the application, because while I was able to find DejaVuSans as "/usr/share/fonts/dejavu-sans-fonts/DejaVuSans.ttf", I did not have much luck with finding Helvetica and Arial.

rbrtjns90, 2025-12-11:
> I still get a floating point error on Linux (Fedora) right after the message "[Main] Entering main loop..."

[image] Ok so I made a special build script and it builds fine for me, maybe the build script is the secret sauce. I'll update the current code to add this build script and you should be good to go. The script is here -> [https://github.com/rbrtjns90/PocketMage_PDA/blob/main/desktop_emulator/build-red-hat.sh](url)

Jonson26, 2025-12-11:
Ok, so it seems that for some reason OLED_MAX_FPS in HOME.cpp is set to 0, which is causing the issue.

Jonson26, 2025-12-11:
Adding `if(OLED_MAX_FPS==0) OLED_MAX_FPS=60;` at line 417 in HOME.cpp makes it work, though this obviously isn't a proper fix.

[image]

Jonson26, 2025-12-12:
I did a fresh pull from your fork.

rbrtjns90, 2025-12-12:
Fixed by setting defaults in globals.cpp. I updated the code. It's two lines in globals.cpp. Lines 36-37 -> int OLED_BRIGHTNESS = 255;  
int OLED_MAX_FPS = 30; 

On fresh installs, FPS was uninitialized before loadState(). I think this was your bug. Thanks for pointing that out.

closed 2026-02-09 by ashtf8