TalismanDesign/PocketMage_PDA #15: Some scripting language for new apps.

gusteivos, 2025-08-13:
It would be good if the firmware supported some scripting language to make it possible to create new applications without being integrated into the firmware.
For this, I recommend Lua as it would not be difficult to integrate it into the firmware and export the firmware interfaces to these applications.

gusteivos, 2025-08-15:
I had never heard of Wrench, looking at the repository it seems to be simple to integrate into the project, but the problem is that the community seems to be small.

ironlungx, 2025-08-30:
wrench seems very cool and its very easy to work with too, here's an example on how to compile code from a file on LittleFS (basically any filesystem):

```cpp
WRState *state = wr_newState();

unsigned char *outBytes; // compiled code is alloc'ed
int outLen;

File f = LittleFS.open("/foo.wr", FILE_READ);

int size = f.available();
char *buf = (char *)malloc(size + 1);
f.readBytes(buf, size);
f.close();

buf[size] = '\0'; // ensure null termination
Serial.println(buf);

int err = wr_compile(buf, strlen(buf), &outBytes, &outLen);

if (err == 0) {
  wr_run(state, outBytes, outLen,
         true);
}

wr_destroyState(state);

free(buf);
```

and adding bindings is also really easy, here's a print function
```cpp
void print(WRContext *c, const WRValue *argv, const int argn, WRValue &retVal,
           void *usr) {
  char buf[128];
  for (int i = 0; i < argn; i++) {
    Serial.print(argv[i].asString(buf, 128));
  }
}

wr_registerFunction(state, "print", print); // execute during initialization somewhere
```

---
reference: https://home.workshopfriends.com/wrench/www/

ironlungx, 2025-11-13:
https://github.com/edubart/minilua/

this seems useful

rbrtjns90, 2025-12-09:
https://github.com/rbrtjns90/PocketMage_PDA/tree/main/Code/AstraLuaApp

closed 2026-02-08 by ashtf8