TalismanDesign/PocketMage_PDA #29: Dynamic OTA loading and partition switching for side-load apps

ashtf8, 2025-08-27:
The ability to side-load apps from the microSD card by loading them OTA into another partition on the ESP32 and then switching partitions to the one containing the app. Switching back to the main partition returns the user to PocketMage OS.

ironlungx, 2025-08-29:
https://mischianti.org/esp32-firmware-and-filesystem-update-from-sd-card-1/

this looks very promising, I'm thinking we have like the following filetree on the SD card:
```yaml
root/
|- apps/
   |- program1.tar  # Maybe we could also have a custom file extension
   |- program2.tar  # because it doesn't matter
```
---

- tar files are basically zip files without compression so that it becomes very trivial for non tech people to just copy one file into the folder. 
- to extract it [this](https://github.com/tobozo/ESP32-targz) looks good plus we could also compress the files but that would complicate things a lot.

---
we should also have some simple JSON payload app metadata and maybe a small image for the program!

```yaml
root/apps/program1.tar/   # extracted
|- firmware.bin
|- metadata.json
|- sprite.bmp    # or whatever format, i am not too sure :/
```

and in `metadata.json`:
```jsonc
{
  "product_name": "Very Cool program",
  "firmware_id": "cool_program",       // maybe we only allow very specific chars
  "version": "2.1.0",                 
  "build_date": "2025-08-29T22:00:00Z",
  "description": "Calculator Premium, with support for division by zero!",
  "author": "Ironlung",
  "checksum": "a3b5c6d7890ef123456789abcdef123456789abcdef123456789abcdef1234567",  // so that we can check if the firmware was transfered correctly and stuff
// more fields can be added 
}
```

---

I will get started on a basic version because this seems like a cool project as it is!
I shall keep y'all updated too

ashtf8, 2025-08-30:
Looks great so far!

ironlungx, 2025-11-10:
@ashtf8 using ELF files would be a much better idea I think, they should be faster because you don't have to reflash and stuff. It should be a lot easier to add custom metadata **before** the ELF bits in the file.

```
[Name][Version][Icon small][Icon Large][Rest of the ELF file]
```

We could parse out the starting metadata without any need for tar/zip. This should make things a LOT faster!

Here's a **rough** sketch I came up with:
https://github.com/ironlungx/elf_loader

I might work on making a better library/wrapper around this

---

I haven't checked what you're using right now but this approach seemed much more straight forward to me